judge2005/EleksTubeIPS #6: Clock freezes randomly

Spider19996, 2024-03-10:
Hey, I installed this firmware on my NovelLife Tube Clock. It works fine but sometimes the clock just freezes for 5 minutes or longer and does nothing. Eventually it will start running again.

Is there any update in the background that hangs and can be switched off or something?

Thanks for any answer

judge2005, 2024-03-10:
What version are you running? You can see it on the info screen?

Also one of my testers has seen that the clock can start to freeze up if his phone is near it! I assume it somehow interferes with the WiFi connectivity.

Either way, yes the weather retrieval is likely to be the culprit as it does retries if it can't connect to the server. The code is multitasking, but retrieving the weather uses a lot of memory which I suppose might interfere with the clock task. If you clear the API key it won't try to retrieve the weather.

Spider19996, 2024-03-11:
Software Rev is 1.0.2
Chip Rev is 3

The wifi of this watch is relatively unstable by nature, even with the default firmware. Maybe that's it too. 

What also disturbs the watch in any case is when you open the web interface and do things there... Changing the clock face is also relatively unstable, but you don't do this every day so this is okay.

I think if it really is the weather, then that's the way it is... I'll try to deactivate it for testing.

Couldn't the queries for the weather be set to certain times, e.g. twice a day and on clock start? Or is this already the case? Unfortunately I am not a programmer, but I have respect for this great work. It definitely enhances the clock a lot.

And thank you for taking the time for me.

Spider19996, 2024-04-18:
Thank you, it's definitely much better. The website is also more stable now and the clock no longer freezes when changing the clock faces. Let's see what the long-term test shows. Sometimes it also worked relatively stable with the old version...

closed 2024-04-18 by Spider19996