judge2005/EleksTubeIPS #8: Display select doesn't seem to be working properly on Si Hai version of the clock

clarksnc, 2024-04-06:
Hi,
This software is fantastic and works fine on two Elekstube versions of the clock I have, however the Si Hai version doesn't seem to drive the displays correctly - it seems to drive all the displays simultaneously so all the displays tick through seconds and then when a more significant digit updates that briefly flashes up across all displays and then the seconds tick up again and so on.

I originally loaded the firmware from nixies.us but have subsequently grabbed the source and compiled on platform.io however I have the same issue

The SmittyHalibut software works correctly and similarly I've built and modified that that on platform.io and by build of that works.

I've tried to track the difference in the source code and libraries but had no joy in finding it, so any assistance greatly appreciated.

Thanks,
Steve

judge2005, 2024-04-07:
I think the problem is in the RTC code. The pins it uses are hard-wired and they clash with the pins used to control the display. It is more complicated than that though. The SiHai uses a different RTC chip than the other clocks and the hardware interface is different. I think a first stab at getting the code to work on SiHai will be to ignore the RTC chip and just use NTP. I will start by committing some code that should do this.

judge2005, 2024-04-07:
> I think the problem is in the RTC code. The pins it uses are hard-wired and they clash with the pins used to control the display. It is more complicated than that though. The SiHai uses a different [...]

I've committed a change that comments out the RTC code for SiHai. If this fixes the issue then we will know for sure that the RTC code is the source of this bug. Please try this and let me know what happens. Note the clock will still work just fine as long as it can connect to the internet because it will just use the NTP time sync.

clarksnc, 2024-04-07:
Thanks.  I'd left the clock with the Smittyhallibut firmware running overnight and coincidentally today I noticed the RTC wasn't working and the NTP wasn't refreshing at a frequency to fix it so I put a quick kludge in to rely on NTP - so it may be that smittyhalibut code has a kind of inverse of the issue in that a bug with the RTC not working meaning their display works - anyway I'll quickly grab your update to test and come back to you.

clarksnc, 2024-04-07:
OK - quickly grabbed the update and I'm pleased to say the display issue is fixed!
Many thanks for a quick response.  Another observation I had with the smittyhallibut firmware, in case it sheds light on a hardware pin clash, is that the  LED backlight in the rightmost column (seconds units) was flickering regularly but it is fine on this code.

closed 2024-04-07 by clarksnc

judge2005, 2024-04-07:
> OK - quickly grabbed the update and I'm pleased to say the display issue is fixed! Many thanks for a quick response. Another observation I had with the smittyhallibut firmware, in case it sheds [...]

Just committed code that should work with the RTC chip on this clock. It compiles, but I have no way of testing it. If you have time could you try this out? If it works you should see the time displayed almost as soon as the clock is turned on. You may need to wait for a NTP sync for the time to be correct though. So I guess, flash the latest, wait for a NTP sync then restart the clock and see if you get the correct time displayed before it manages to do another NTP sync.

Maybe clone this repo to another location so you can easily go back to the previous version if it screws up, though I would hope that even if the new code doesn't correctly read the time from the RTC chip, it would still be fine after an NTP sync.

clarksnc, 2024-04-07:
Hi,  thanks - Sunday evening UK time, but I'll try to test early in the coming week.

clarksnc, 2024-04-07:
UPDATE:  Sitting in front of the TV with my laptop I thought I'd get this up and running tonight!

Got it built OK, but I think use of the RTC exposing a hardware issue of some kind on this board.  The display issue I originally reported is resolved but I encountered a number of other issues.  Firstly the onboard menus that show up initially seemed very clunky and almost impossible to dismiss - as if the button presses aren't being recongnised.  The rightmost backlight LED periodically flickers (as observed as per my previous comment on the smittyhalibut code); but strangely wifi is almost impossible to connect with it failing to initially cinnect and when it finally did  dropping after a short time, and was not able to get weather (which I'd also set up as a test with a valid API).  Without putting more debug serial prints in I couldn't honestly say if the RTC is working either!

I reverted to your earlier quick fix build reliant on NTP rather than the RTC and was up and running and solidly connected to WiFi  (with the device in the same location) with weather again within moments and no flicker on the rightmost backlight LED.

judge2005, 2024-04-07:
OK. Well I’ll just turn off the RTC for that clock then.