judge2005/EleksTubeIPS #9: MQTT "Error loading Page"

Spider19996, 2024-04-16:
Hey, I just installed 1.5.2. When I try to open the MQTT page it just says "Error loading page". Is it possible to view a log file or something like that to attach it to this post?

Spider19996, 2024-04-16:
I Just updated from the nixies.us Website and did a "Firmware only" update. It reset all the settings but I kept the clock faces that way...

judge2005, 2024-04-17:
You will need to update the file system to get the MQTT page :(

Spider19996, 2024-04-17:
Thanks for your answer... So I need to do a "full install"? I tried that and it only reports "Downlading firmware bootloader.bin failed: 404". So I tried to revert to Stock and now Im stuck on the default firmware... I can't install the replacement firmware. :(

Spider19996, 2024-04-17:
So I did a full flash with the Adafruit WebSerial ESPTool and now it works... Again thanks for your work and help! I leave the issue open because of the firmware download problem from the nixies.us website. But can otherwise be closed :)

judge2005, 2024-04-18:
I fell foul of how wordpress organizes its uploaded files, so it couldn't find a few of the unchanged files that were in the manifest. Fixed now. I'll close this.

judge2005, 2024-04-18:
Fixed web site

closed 2024-04-18 by judge2005