judge2005/EleksTubeIPS #12: "Failed to delete file" with incorrect file format

Spider19996, 2024-04-21:
Hey, just found another bug...

Uploading the wrong file format will result in the file being stuck there forever, as only the error "Failed to delete file" is displayed when deleting it. Example: It is possible to upload a *.tar file but never delete it again. (Expected file format *.tar.gz)

judge2005, 2024-04-21:
How do you upload files that don't end in .tar.gz? On my system I can only select files that end in .tar.gz

Spider19996, 2024-04-21:
That's right, only *.tar.gz files are displayed via the file picker built into Windows... I always dragged the files from the Explorer into the file upload dialogue and then clicked on Upload.

On an Android smartphone, you can also upload everything using the built-in file picker.

judge2005, 2024-04-21:
Workaround is to re-flash just the file system binary.
- You can find the binaries in the [releases section](https://github.com/judge2005/EleksTubeIPS/releases). You can find the release number in the info page of the web GUI - it is called _Software Rev_
- There are multiple ways that you can re-flash just the file system binary. Pick one from [here](https://github.com/judge2005/EleksTubeIPS/wiki/Installation#installation).

closed 2024-12-07 by judge2005