judge2005/EleksTubeIPS #14: Screens in reverse order

eskudo12791, 2024-05-23:
Apologies ahead of time for my lack of familiarity. I purchased the "RGB glow tube clock IPS" and loaded the "Punkcyber" firmware via the web installer but the screens are in reverse order. Is this not the appropriate firmware for my device?

judge2005, 2024-05-23:
Try the regular Elekstube firmware https://www.nixies.us/projects/elekstubeips-clock/elekstube-ips-v1/

eskudo12791, 2024-05-23:
judge2005, thank you for the reply. That version seems to work but is unstable, when I change clock faces they will start to load but then go blank, to make them come back I have to cycle the power and then screen 3 will randomly just go blank. Outside of Punkcyber being backwards the display was stable. Possibly I have a temperamental board?

judge2005, 2025-04-11:
> judge2005, thank you for the reply. That version seems to work but is unstable, when I change clock faces they will start to load but then go blank, to make them come back I have to cycle the power [...]

Maybe this other issue will help? #25

closed 2025-04-11 by judge2005