judge2005/EleksTubeIPS #16: Version 2.5 of OpenWeatherMap API has been closed.

wsmmsh, 2024-07-03:
OpenWeatherMap API 2.5 has been closed in June 2024. Now I found EleksTubeIPS weather doesn't work.
If the OpenWeatherMap module uses API 2.5 as if it does, it won't be working much longer. 
Could you get it upgraded? Thanks verymuch.

judge2005, 2024-07-05:
I don’t see anywhere it says that, in fact the web site still promotes 2.5 along with 3.0. Do you have a link?

wsmmsh, 2024-07-06:
> I don’t see anywhere it says that, in fact the web site still promotes 2.5 along with 3.0. Do you have a link?

Look here https://openweathermap.org/api/one-call-api
I can't apply the version 2.5 of their API now.

judge2005, 2024-07-07:
I don't use the 2.5 one all API. in their examples they have:

https://api.openweathermap.org/data/2.5/**onecall**?lat={lat}&lon={lon}&exclude={part}&appid={API key}

I checked a call that I use and it still works (note the word **onecall** is not in the URL). i.e

https://api.openweathermap.org/data/2.5/weather?lat=43&lon=-72&appid=<appid>&units=imperial

Their migration page explicitly says:

"Please note that no other products are affected by this change, including other APIs that have ‘/2.5/’ section in their URL."

judge2005, 2024-07-07:
This is the forecast API I use:

https://openweathermap.org/api/hourly-forecast

This is the current API I use:

https://openweathermap.org/current

wsmmsh, 2024-07-09:
Sorry, I didn't read the website carefully. I have fixed the problem now. Thanks very much.

closed 2024-12-07 by judge2005