judge2005/EleksTubeIPS #19: Occasionally stops showing time and responding

MennoLabs, 2024-09-30:
Sometimes the clock stops for 30 - 90 seconds.
It will stand still, eventually jump to the current time.
Web interface also not available at those moments.
Can't ping the clock either.

judge2005, 2024-10-01:
This is most likely because of Wi-Fi issues - the placement of the ESP32 module in the clock is not great from a radio perspective, which can make the Wi-Fi connection susceptible to interference. A friend had this when he was charging his phone next to the clock for example. Another possibility is that the power adapter it is plugged in to is not providing enough power, so try a different one.

MennoLabs, 2024-10-01:
I guess with WiFi enabled the power requirements have gone up considerably compared to the original firmware.
It is now connected to a max 0,5A power supply.
A bit of Googling digs up:
"It has also been observed that the chip draws more than 790 mA at times, particularly when both WiFi and Bluetooth are used simultaneously."
Will find a beefier power supply and try that.
Thanks!

closed 2024-12-07 by judge2005