judge2005/EleksTubeIPS #20: Clock won't accept my timezone

MennoLabs, 2024-09-30:
I live in a timezone with a long definition:
CET-1CEST,M3.5.0,M10.5.0/3
The web interface does not accept this as input.
I get "Error loading page" when I enter the time zone and hit enter. Time won't be set to local time.
For now I use <+02>-2, which gives me the correct time, but I guess I'll have to adjust for DST manually.

judge2005, 2024-10-01:
The clock should be able to handle a string up to 60 characters. I wonder if there is a bug in the posix TZ library. Try leaving off the last two characters '/3'. Those characters tell it to adjust to winter time at 3am. If you leave them off, it will happen at 2am instead.

MennoLabs, 2024-10-01:
Weird:
Enter CET-1CEST,M3.5.0,M10.5.0
Get "Error loading page" again.
Reload page.
Field now shows CET-1CEST,M3.5.0,M10.5.0/3
Consider it solved...
Thanks!

closed 2024-10-01 by MennoLabs