judge2005/EleksTubeIPS #21: NovelLife SE issue with "always off"

hthiel75, 2024-11-02:
I installed the firmware via https://www.nixies.us/projects/elekstubeips-clock/elekstube-ips-v1-2/, and everything went smoothly. Unfortunately, the function always stays in the "off state," meaning that no matter what I set, it only shows what I configured as the "off state" in the web interface (off, dim, matrix).

It’s possible that I have a hardware issue. Could someone configure the firmware for me without this "display on" setting so that the display stays constantly bright?

judge2005, 2024-11-03:
Please post screen shots of all the web pages in the GUI.

hthiel75, 2024-11-03:
[image]
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

judge2005, 2024-11-03:
It's behaving as if it thinks the power button is being held down. There could be one of two things wrong here:

- The polarity of the power button isn't what I think it is. If this is the case then if you just long press the power button the display should come to life (except if you have the 'off state' set to dim, in which case it'll stay dim, so I recommend you set the off state to blank before you try this, that's a bug). If this works you are all set. Just remember to do this if you power cycle the clock. Though I would probably release a version that fixes this.
- The power button is being read as 'pressed' no matter what state it is actually in, in that case read on:

I tried a test where I unplugged my clock, held the power button down and then plugged the clock back in. The display came up until it had initialized and then went off.

Try this:

- Set the 'off state' to blank
- Set the 'Display on' control to have the clock **off** for whatever time it is now where you are. In my testing (where I hold the power button down), this first attempt to change the on/off times fails - the control jumps back to whatever it was. If that happens, do it again and it should take.
- Set the 'Display on' control to have the display on.

If I am correct, at this point the clock should turn on.

Let me know if either of these work or not.

hthiel75, 2024-11-03:
The Novellife SE has no power button—or at least, I wouldn’t know where to find it on my version :-)

I tried the method you described, and lo and behold: it works and has been running without issues for the last two hours. Perfect, thank you !

Martinius79, 2024-11-03:
Hello again @hthiel75!

I am curious now, which version of the NovelLife clock do you own?
Can you make a photo and upload it?

I have the following: NovelLife: SE - Spectrum-V2.0
It has not Buttons, just the APDS9960 gesture sensor in front.
[image]

And I also had some trouble to get out of matrix rain screensafer. I needed to switch the settings a bit arround, to get the clock to show the undimmed digits of the actual time.

So maybe it will also work for you:
Screenshot of my settings:
[image]

I switched to "six" first, then "date" and back to "time", then "Dim" and to "Blank", then I turned the "wheel" until it went "gray" nearly completly, like 23-24, then I turned it back, until it went blue again and shows 0-24.

That makes the clock at least show the time in undimmed digits for me...

So, the question is, what are you hoping to achieve with your clock?

judge2005, 2024-11-03:
Now I'm curious. These are the clocks with gesture control. Does using gestures do anything with my code?

Martinius79, 2024-11-03:
No, the gesture sensor is not doing anything, as long it is not initalized.

But there are (or was) two kinds of Novellife clocks on the markets. 
Novellife Pro (with buttons), see https://www.novellife.net/productinfo/941234.html
and
Novellife SE (without buttons, with gesture sensor), see https://www.novellife.net/productinfo/941237.html

Different hardware revision of the SE exists, like "Spectrum-V2.0" or "V2022-Gesture-V2.1", but they seem do be compatible with each other.

The Pro version is long time out of sale, I saw it available last time in Dezember 2023. So I guess it is irrelevant, as long as nobody demands support for it :)

The problem here can be, that you are using the "default" button pin layout. Some of the pins are used differently on the SE (34 and 35), so I prefered to "kill" the button pins by setting them to an always high pin (GPIO 3 i.e.). I did this in my implementation for the gesture sensor support on the ElekstubeHAX project.

I modifed your code in the fork a bit and it is nearly perfectly working now. 
see https://github.com/Martinius79/EleksTubeIPS

I modfied the GLOBAL_DEFINES.h
```c_cpp
  // No Buttons on SE verion -- gesture sensor not included in code 
  #define BUTTON_LEFT_PIN (3)
  #define BUTTON_MODE_PIN (3)
  #define BUTTON_RIGHT_PIN (3)
  #define BUTTON_POWER_PIN (3)
```

The only problem I have now, is that the "night time dimming" is not working as expected. If I switch to "Dim" instead of "Blank" or "Matrix", the displays stay dimmed all the time. The set time range seems to be ignored. For the other two modes, it is working.
I wasn't able to watch in your code that deep in the moment to check, what is going on there...

And I also noticed, that with the actual espressif framework and the actual PIO espressif integration, the code consumes to much DRAM...I have no idea, how to fix this... :(

judge2005, 2024-11-03:
Thanks for doing that. I'll change the implementation to disable the inputs - or produce two binaries.

The _Dim_ settings has a bug, which I will fix at some point.

It sounds like you got it running, so I'm not sure what you mean when you say it consumes too much DRAM - that implies that something doesn't work?

Martinius79, 2024-11-03:
It happens, if you use the actual versions of the PIO espressif imtegration.

platformio.ini:
```
...
[env]
platform = espressif32@ ^6.9.0
board = esp32dev
framework = arduino
platform_packages = framework-arduinoespressif32 @ 3.20017.0
...
```

```
---/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/elekstubev1/firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
---/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
---/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
---/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 864 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\elekstubev1\firmware.elf] Error 1
```

I forcibly deactivated the weather stuff to be able to build it with that version of the framework.

BTW: I am really impressed with what you are doing with the ESP32! I wish I had your coding skills for microcontrollers and C++ :)

hthiel75, 2024-11-05:
Thank you all for the effort you put in. Thanks to the workaround, I now have a perfectly functioning clock.

judge2005, 2024-11-24:
v1.6.0 should work without this workaround