judge2005/EleksTubeIPS #23: littlefs-1.6.0.bin too big for flashing

gans-spb, 2024-11-30:
Hi, 
[v1.6.0](https://github.com/judge2005/EleksTubeIPS/releases/tag/v1.6.0), seems [littlefs-1.6.0.bin](https://github.com/judge2005/EleksTubeIPS/releases/download/v1.6.0/littlefs-1.6.0.bin) is quite big (6.44 MB) for the ESP 4GB.

```
esptool.js
....
0x180000 littlefs-1.6.0.bin
...
Wrote 6750208 bytes (1395870 compressed) at 0x180000 in 79.328 seconds.
File  md5: fc00dba853640ff4a3f76f200ca0f270
Flash md5: 225ad1e6fe8acbf519f664cecfb3847f
Error: MD5 of file does not match data in flash!

esptool.py v4.7.0
Chip is ESP32-D0WDQ6 (revision 1)
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
Device: 4016
Detected flash size: 4MB
```

thanks for the great soft!

judge2005, 2024-12-01:
This should be fixed now. There are two littlefs images:

- littlefs8MB-1.6.0.bin - this is for the ipstube clocks only, which have more memory than the others
- littlefs-1.6.0.bin - this is for every other clock.

gans-spb, 2024-12-01:
I appreciate your quick response! Uploaded Ok.

Quickly checked this release and only a few things were misinterpreted:
1. WebUI -> Clock -> Off State -> Dim - immediatly apply Dim effect, didn`t  switch on-off like Matrix; looks like bug
2. WebUI -> Clock -> Display On  -> Time Selector Ring. It is not obvious for me that this is sleep time selector. May be a better name "On-Off display time (blue-on)", and vertical alignment (turn 90 cw, "0" up, like normal clock).

Also, I have a trouble with last 6th LCD digit. The image corrupted, especially when picture is changing, while Matrix screensaver or seconds change. Sometimes 1st ant 5th LCD modules corrupted pictures too, colours and pixels. Those leads to black screenc after. Not sure this is a software bug. My clock is one of the first variant, may be a hardware bug. Didn`t find schematics. Spend some time in analyzis of this behaviour.

judge2005, 2024-12-05:
Thanks for the feedback. If you switch to the date display using the date button in the GUI, it should allow you to pick the date format.

The screen corruption is interesting. If you can find a more powerful power supply try that. If you're OK with building the binary yourself, you can drop the SPI frequency in the platformio.ini file. If not, I could build one but I would prefer not to upload it nixies.us - are you OK flashing the binary using one of the options I describe in the wiki.

closed 2024-12-16 by judge2005