judge2005/EleksTubeIPS #24: most right screen fails to initialize - IPStube

Trigger-4622, 2024-12-14:
Installed firmware 1.6.2 from the web installer.
(As for my understanding the most recent 1.6.2).

The right most screen fails to initialize, in stock firmware it works just fine.
Images attached bellow.

Feel free to msg me ill try to pinpoint the issue with my hardware, Thank you so much for your effort with this firmware!

[image]
[image]

judge2005, 2024-12-14:
Well that's interesting as I don't think that firmware should work at all with this clock (I received my ipstube clock today and it didn't work with mine!). I've uploaded a new version to https://www.nixies.us/projects/elekstubeips-clock/elekstube-ips-v1-2-2/. I noticed that I have to put my clock in boot mode and that I can't configure the WiFi directly from the web page - I've updated the instructions accordingly. Let me know if this works.

Trigger-4622, 2024-12-14:
Will give the new one a chance right now!
It took me quit a few tries to get it to work with any firmware at all. i dont know what changed but after a few tries it just worked... really strange.... 
If you want to have a look i found another guy who made a working firmware that initializes all of the screens, but its premature lacking features other than the clock and mqtt non the less awesome work too. 

Also.... I kind of received my clock with no firmware installed at all... any chance you could please upload a copy of the original firmware?

Trigger-4622, 2024-12-14:
Well, just installed 1.6.3 and it woorks!
Just remembered, i reuploaded the bootloader, maybe it had somthing to do with it?
let me know if you need me to do anything to help with bug fixing, will be more than happy to!
[image]

judge2005, 2024-12-14:
I've attached my backup of the original firmware to this comment.

Kind of jealous of the underlighting!
[ipstube.bin.zip](https://github.com/user-attachments/files/18137048/ipstube.bin.zip)

closed 2024-12-16 by judge2005