judge2005/EleksTubeIPS #25: TFT_ips most right figit fail

gans-spb, 2025-01-15:
I've got some investigaton and will describe a **TFT image corruption** bug, that I have.
EleksTubeV1 old hardware, firmware 1.6.4. This bug I didn`t remember at the original firmware.

Corrupted picture at the _most right indicator_ and sometimes other digits, sometimes leads to a de-initialized black screen. This bug well appeared on the fast picture changes: matrix screensaver, file loader ring, seconds changes. The bug is more obvious when SPI_FREQUENCY changes to 20MHz and less.

HW signals seem to be OK on the oscilloscope. May be CS is to close to the Data, but additional delays in the programming code ChipSelect::update() didn`t resolve bug. TFT modules permutation, redoldering, PCB washing or heating, lab PSU, flash full erase - no changes.

I`ve write small test (see elex_test.cpp code below) using fillRect() and setDigit() APIs, at the beginning of setup(). It works OK for 40MHz SPI only, and the bug began at lower frequency. 

I don't have any idea what causes this. This can be inside TFT_eSPI library, or RTOS cross-thread correlation, or somewhere in the hardware. Maybe this is only my personal bug; in this case, it can be closed.

[elex_test.cpp.txt](https://github.com/user-attachments/files/18429982/elex_test.cpp.txt)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

judge2005, 2025-01-16:
I haven't seen any issues like this since my early days developing this code and IIRC they weren't code related. A couple of things to try:

1. Reseat the display that has the problem.
2. Use a power supply with a higher rating.
3. Move the clock to a different location.

This last one because the clock isn't designed to use the WiFi radio, so connectivity can be flaky - someone who acted as a beta tester for me had issues with the clock just randomly rebooting. We finally figure out that it only happened when he was charging his phone near the clock.

Let me know how it goes

gans-spb, 2025-01-31:
Victory! Spectacular HW error!

I have about 5 years old Elekstube clock, and seems **TFT flat cable invisibly damaged**, dry out, lost contact. And those "drought" flat cable TFT extremely influence to the neighbours through IPS line by damage the signal. I mechanically killed one of TFT while experiment, by the new one, change - and voila! All 6 digits works almost right.

Conclusion. If your clock is old, and has the TFT image corruption - try to find a "bad apple" and change TFT to the new one.

Ticket can be closed, this bug may be announced in the FAQ. Thank you.

[image]

closed 2025-02-01 by judge2005