judge2005/EleksTubeIPS #30: IPSTube H401 - Loss of some features

Phlavio-SPR, 2025-02-01:
Hello.
First of all, congrats for author efforts to create a very cool firmware replacement.
In almost all respects, it is much better than the stock one.
But there are some missing stock features that prevented me to keep this firmware on my clock, so I switched back to the original.
IPSTube H401 hardware provides ability to adjust lcd backlight brightness and also has an integrated RTC with battery backup. Both of them doesn't seem to be supported by this alternative firmware.
Is it possible to implement a full support for that?
Thank you!

judge2005, 2025-02-02:
It does support the RTC chip and there is a brightness slider on the main clock screen. I guess I need to update the wiki. It currently adjusts the brightness by dimming the images rather than adjusting the backlight.

[image]

In addition you can use homeassistant to adjust the brightness using a light sensor.

Phlavio-SPR, 2025-02-02:
Thank you for a prompt answer. And congratulations again for your firmware.
Glad to know that RTC chip is supported already.
Regarding the brightness thing, depending where the clock stands, there is a huge light leakage that make it looks ugly when it is dark.
Would be possible to divert the brightness slider to a PWM control over GPIO4 where the backlight driver is connected to?

drevel123, 2025-02-19:
Hello,
I just got an IPSTube as well and immediatly flashed your firmware.
First of all I have to thank you for making this, too - I wouldn't have even bought the clock without knowing about your great firmware. I love being able to control the clock via a simple webinterface (without any apps or cloud services) and even access it via my homeassistant server.
Regarding the brightness slider: I cannot see it on the main clock screen (see screenshot below).
Another stock feature I am missing tho is the possibility to display static images. Is it maybe possible to implement an option to upload five pictures to the clock to display regardless of the time like the 'LOVE'-watchface in the stock firmware?

Missing those two features won't make me revert to the stock firmware, but with those the clock (or 'mini picture frames' when using static images) would be simply perfect in my eyes.

[image]

judge2005, 2025-02-20:
Looks like I didn't upload the latest version to nixies.us. I'll try to get around to that this weekend.

I've been banging my head against another project for a while, so haven't had time to work on this one.

Space is tight in there for any additions. I might have to consider dropping the matrix screen saver and perhaps replace it with a slide show screen saver?

judge2005, 2025-02-21:
OK. I've updated the installation page with the latest firmware. Note that this will lose your current settings (because I added a new one) so you'll have to set it up how you like it again :-(

drevel123, 2025-02-21:
Thank you for your very quick response and updating the installation page, I installed the latest firmware and I've got a working brightness slider, now.
A slide show screensaver with images I can choose would be even more awesome than static images in my opinion.

I completely understand that you are busy with other projects. I truly appreciate the effort you've put into this one and I'm more than happy to wait for any updates - no matter how long the time :-)