judge2005/EleksTubeIPS #31: IPSTube - some inconsistencies

ElToberino, 2025-02-07:
First of all thanks a lot for making this software also available for the IPStubes!
Having used and tested your firmware for a while I noticed some things - I don't know if they are intended or if they are bugs.

1. Switching off the clock with a long button press works fine. Switching on again with a short press works, but some seconds later the clock goes into the off state again. Switching on with along press works fine - the clock remains in the on state.
2. Off State: The background leds are off in the modes "Blank" and "Matrix". In "Dim" mode they remain on. Is this difference intended?
3. In Weather Mode (only) with chosen Off State "Blank" the leds are turned off (as intended), but the displays remain on.

Furthermore some suggestions for a future version:
Wifi: 
The possibilty to set a static IP would be great. It would also make sense to enable accessing wifi manager via web interface to change/delete wifi parameters.
Web-Interface: 
It would be nice to have a button for turning on/off the clock via web interface and an additional mode which cycles through the three mode time, weather and date.

Thanks again for your work!

judge2005, 2025-04-11:
Point one is the intended functionality. The idea is/was that you might want to briefly see the display when it is in the off state.

The other two are bugs:

- LED brightness should be in sync with the display brightness
- I mean yes, the display should be off for all displays!

judge2005, 2025-04-12:
> The possibilty to set a static IP would be great. It would also make sense to enable accessing wifi manager via web interface to change/delete wifi parameters.
AFAIK static IP is the down to the router - i.e. you configure a static IP for this device in the router. The device name should be the same as the host name.

I've added a Network page to the GUI in v1.6.7 - you can change the hostname directly (the clock should reboot if it takes effect) and you can manually enable the access point so you can re-configure the WiFi from there.

judge2005, 2025-04-12:
> Point one is the intended functionality. The idea is/was that you might want to briefly see the display when it is in the off state.
> 
> The other two are bugs:
> 
> * LED brightness should be in sync with the display brightness
> * I mean yes, the display should be off for all displays!

Fixed in v1.6.7

judge2005, 2025-04-19:
Leaving this open for these enhancements:

Web-Interface:
It would be nice to have a button for turning on/off the clock via web interface and an additional mode which cycles through the three mode time, weather and date.

judge2005, 2025-11-29:
I haven't added a mode for cycling through the time, but I have added support for changing the mode to the HA integration, so it can be automated that way.

closed 2026-11-07 by judge2005