judge2005/EleksTubeIPS #32: Help converting images for the clock

pmsobrado, 2025-02-13:
What the title says. I downloaded some themes that are not on the proper format, so I tried to convert them with this command with your python script:

`python batch_convert_images.py *.png -p 24 -s -o bmp`

However, the result is just a black screen on the clock. If I try uploading two or three times my conversions, there's a point where the other working themes also stop working, maybe a memory issue or something?

Here I attach the source images and the result images. Could you tell me what's wrong with them?

[result.tar.gz](https://github.com/user-attachments/files/18783014/result.tar.gz)
[source.zip](https://github.com/user-attachments/files/18783013/source.zip)

Thanks!

judge2005, 2025-02-13:
It is apparently compressed wrong. What command did you use to compress it? This command would work: _tar cvzf result.tar.gz *.bmp_ or _tar cvf result.tar *.bmp_ followed by _gzip result.tar_.

I've attached a result file with the images converted to 16 bit rather than 24 bit. The displays are 16 bit so there's no point them having more bits per pixel.

[result.tar.gz](https://github.com/user-attachments/files/18790998/result.tar.gz)

pmsobrado, 2025-02-24:
@judge2005 sorry for the late response, I've been ill. Your file doesn't work either, but that's OK, I just wanted to know if I was doing something wrong. Maybe my clock is a bit faulty at this point. As I told you, if I try uploading two or three times my conversions, there's a point where the other working themes also stop working.

judge2005, 2025-02-24:
> @judge2005 sorry for the late response, I've been ill. Your file doesn't work either, but that's OK, I just wanted to know if I was doing something wrong. Maybe my clock is a bit faulty at this [...]

Reinstall the firmware and select the erase option, then try my file again. There is probably just some bad data lying around in the filing system.

pmsobrado, 2025-02-24:
Yes, that is what I did last time, I'll try again and let you know.

pmsobrado, 2025-02-27:
Sorry @judge2005, no dice, I reflashed the firmware with the erase option but your tar.gz still doesn't work. Don't worry though, that's ok.

LuhiFox, 2025-03-31:
> Reinstall the firmware and select the erase option, then try my file again. There is probably just some bad data lying around in the filing system.

Your file dosen't work for me either. I'm too stupid to get the Python script to work

LuhiFox, 2025-04-01:
> I did use the python script, but it didn't work. Also, when downgrading the imgs depth, the quality was very bad.

Ok I figured out. I don't used the Python Script. I used Irfan. Shrink your image to the format, in this step, you can set the color depth to 8 bit 256 colors. With 16 bit, it doesn't work for me. But 8 are ok.
After that, save your image as a bmp file. In Windows: select all images (minimum 0-9) right click, compress to, more options. Then archive type: Tar(GNU), compression method: Gzip. And this file is ready to upload and this method worked for me

LuhiFox, 2025-04-01:
Alternatively, I can do it for you. Just send me your  pictures.

pmsobrado, 2025-04-02:
@LuhiFox you have them on the first message on this issue :)

LuhiFox, 2025-04-03:
> [@LuhiFox](https://github.com/LuhiFox) you have them on the first message on this issue :)

[results.tar.gz](https://github.com/user-attachments/files/19590306/results.tar.gz)

Try this

pmsobrado, 2025-04-04:
Yes, it worked, thanks! So the trick is to use Irfan and compress with Windows?

LuhiFox, 2025-04-04:
> Yes, it worked, thanks! So the trick is to use Irfan and compress with Windows?

Yes, this method worked for me.