judge2005/EleksTubeIPS #33: MQTT does not connect to Homeassistant Mosquitto

neutrino220, 2025-02-23:
Software Ref: 1.6.5

After entering IP, Port, User and Password, the clock reboots and IP flied is empty after booting and refreshing config-page.

Mosquitto log:
2025-02-23 22:27:55: New connection from 172.30.32.2:53804 on port 1883.
2025-02-23 22:27:55: Client <unknown> closed its connection.
2025-02-23 22:29:55: New connection from 172.30.32.2:42602 on port 1883.
2025-02-23 22:29:55: Client <unknown> closed its connection.

judge2005, 2025-02-27:
Yes indeed! I don't even get that far. Put in the IP address and boom. I will get back to you.

judge2005, 2025-02-27:
OK. All fixed. I will make another release within the next few days.

judge2005, 2025-03-01:
Fixed - I have created a new release (v1.6.6). Install in the usual way.

closed 2025-03-01 by judge2005