judge2005/EleksTubeIPS #34: Your ESP32-S2 board is not supported

yt2008, 2025-03-16:
My Xunfeng IPS Clock is bricked (display blank screen now...)
When I try to flash [EleksTube v1 clock](https://www.nixies.us/projects/elekstubeips-clock/elekstube-ips-v1/) (which includes the Xunfeng IPS clock) 
It shows "Your ESP32-S2 board is not supported"

judge2005, 2025-03-16:
Sounds like they switched to a different processor. I might be able to produce a binary for an S2, but I would have no idea what else might have changed.

What do you mean _It shows "Your ESP32-S2 board is not supported"_? What is _it_ here? That implies that my firmware was not uploaded so I assume that if you just power-cycle your clock, it should be fine.

yt2008, 2025-03-16:
> Sounds like they switched to a different processor. I might be able to produce a binary for an S2, but I would have no idea what else might have changed.
>
> What do you mean _It shows "Your ESP32-S2 [...]

Hi Paul, 
Thanks for a very quick reply.
I had a look at the panel of my `Xunfeng IPS Clock`, yeah, it does use `ESP32-S2-ROOM`. I am not sure about other Xunfeng products,  tried to research it but not find much information online. 
So `It` means, in this page [EleksTube v1 clock](https://www.nixies.us/projects/elekstubeips-clock/elekstube-ips-v1/), after I connect the USB to my clock, I clicked `Full Install for EleksTube IPS V1`, then I choose the connection and check `agree to wipe data`, then the next pop-up shows this message. Another question is what do you mean `power-cycle`? Sorry I am a newbie for this thing, and I have bricked it, it now boot without anything showing on screens, I might need a bit knowledges for this.

judge2005, 2025-03-17:
Unfortunately it sounds like you need to upload the original firmware. You can see if xunfeng have a web site that you can download it from, or you can try to find another owner who could dump the original firmware for you.

I could also build my firmware for that processor, but there's no guarantee it would work because the pin assignments are probably different.

My original assumption that my firmware would work was based on comments for other replacement firmware that I can no longer find.

By 'power cycle' I just mean turn it off and then turn it back on.

yt2008, 2025-03-18:
[elekstubeipsv1.zip](https://github.com/user-attachments/files/19311444/elekstubeipsv1.zip)
I tried to search if there is any Xunfeng website but unfortunately did not find much, this is the firmware I backed up, i tried to flash it back but not work...

Here is a photo of the chip, `ESP32-S2 WROOM`

[image]

I tried power off but not work...

judge2005, 2025-03-18:
Interesting. Looks like if I can get my firmware built for the S2, it should work. I tried building for the S2 but it seems like there is less of a certain type of memory, so I will need to play with that. OTH I tried flashing your binary to an S2 I have lying around (not a clock) and got this, so my question for now is when you tried to flash this yourself, what exactly happened:

```
> esptool.py -p /dev/cu.usbmodem01 -b 230400  write_flash 0x000000 elekstubeipsv1.bin

esptool.py v4.8.1
Serial port /dev/cu.usbmodem01
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v1.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 80:65:99:e1:84:c4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x003fffff...
Compressed 4194304 bytes to 1308493...
Wrote 4194304 bytes (1308493 compressed) at 0x00000000 in 28.3 seconds (effective 1183.9 kbit/s)...
Hash of data verified.
```

yt2008, 2025-03-18:
When I try flashing, I got this shown on my console:

`esptool -p COM3 -b 230400  write_flash 0x000000 .\elekstubeipsv1.bin
esptool.py v4.8.1
Serial port COM3
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 7c:df:a1:01:df:d8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x003fffff...
Compressed 4194304 bytes to 1308493...
Wrote 4194304 bytes (1308493 compressed) at 0x00000000 in 59.4 seconds (effective 564.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

judge2005, 2025-03-18:
That looks like it worked, so what doesn't work after you have done this?

yt2008, 2025-03-18:
[image]

After this step, It is successfully flashed, but still bricked... I can see the screens are on but display nothing, click any button also does not have any response... It became to the initial brick status I made, I initially bricked it not because of flashing firmware, just by changing some configs via the button.

[image]

Then I flash this `data.bin` that I uploaded, and go to this status mentioned above, no display for the clock, and the menu is upside down mirrored, I had another look at the arguments of this step's, it just flashed the data part of the storage

this step the console is like this:
esptool --chip esp32-s2 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq
 40m --flash_size detect 0x290000 .\data.bin
esptool.py v4.8.1
Serial port COM3
Connecting......
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 7c:df:a1:01:df:d8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00290000 to 0x003fffff...
Compressed 1507328 bytes to 211197...
Wrote 1507328 bytes (211197 compressed) at 0x00290000 in 7.2 seconds (effective 1685.8 kbit/s)...
Hash of data verified.

judge2005, 2025-03-19:
Try the attached binary. Its the first time I've produced a single binary image, so I may have messed it up. It doesn't include weather, MQTT or the screensaver for space reasons. Obviously it might not work anyway - it could take us a while to fix any issues.

[xunfeng.bin.zip](https://github.com/user-attachments/files/19330809/xunfeng.bin.zip)

yt2008, 2025-03-19:
flashed `xunfeng.bin`, not work, no display, the screen is not on

[image]

judge2005, 2025-03-19:
Do you get a WiFi hotspot from it that you can connect to? That would let me know if it is a display thing or whether the whole thing isn't working.

yt2008, 2025-03-19:
No, I check from my phone, no wifi hotspot

judge2005, 2025-03-20:
Try this one 

[xunfeng.bin.zip](https://github.com/user-attachments/files/19355607/xunfeng.bin.zip)

yt2008, 2025-03-20:
Just flashed, it is the same, no display, no wifi hotspot

judge2005, 2025-03-21:
OK. Try this one :).

[xunfeng.bin.zip](https://github.com/user-attachments/files/19377541/xunfeng.bin.zip)

judge2005, 2025-03-21:
> Just flashed, it is the same, no display, no wifi hotspot

Try this one... 

[xunfeng.bin.zip](https://github.com/user-attachments/files/19377551/xunfeng.bin.zip)

yt2008, 2025-03-21:
> > Just flashed, it is the same, no display, no wifi hotspot
> 
> Try this one...
> 
> [xunfeng.bin.zip](https://github.com/user-attachments/files/19377551/xunfeng.bin.zip)

Hi Paul,
Just flashed, it is the same, nothing displayed, cannot find Wifi hotspot either

judge2005, 2025-03-22:
And again. I disabled some pin assignments that I think might be interfering with with the ESP32 S2.

[xunfeng.bin.zip](https://github.com/user-attachments/files/19399568/xunfeng.bin.zip)

yt2008, 2025-03-22:
> And again. I disabled some pin assignments that I think might be interfering with with the ESP32 S2.
> 
> [xunfeng.bin.zip](https://github.com/user-attachments/files/19399568/xunfeng.bin.zip)

Good job Paul, the wifi hotspot is shown in my list!

[image]

judge2005, 2025-03-22:
I have no idea how that PCBWay image works. The I/O pins that the S2 chip uses are completely different than the ones the PCBWay clock uses. The only way forward with this is if you can trace all of the pinouts from the S2 module to the various other chips on the board. Even then it could be a long process of trial and error as I don't have one of these clocks to play with.

I feel like your best step forward would be to try and find someone who can read the binary from another clock for you and/or buy an ipstube clock and flash my firmware (they were around $30 on Aliexpress last time I looked).

I've attached a binary that works on WiFi, but it isn't much use as it can't drive the displays - attempting to do that on the S2 board I have  borks the WiFi access point. Probably worth pointing out that it assume things like buttons, backlights and RTC chip are also mapped to certain I/O pins that are most likely also wrong.

[xunfeng.bin.zip](https://github.com/user-attachments/files/19403575/xunfeng.bin.zip)

yt2008, 2025-03-23:
> I have no idea how that PCBWay image works. The I/O pins that the S2 chip uses are completely different than the ones the PCBWay clock uses. The only way forward with this is if you can trace all of [...]

I flashed this attached one, the wifi config is working, you are right, it still displays nothing.
The clock was a gift but unfortunately we cannot save it. 
I had a look at Aliexpress, I believe [this](https://www.aliexpress.com/item/1005007197379815.html) could be the one you mentioned? 
Thanks for your help Paul.

judge2005, 2025-03-23:
This is the one I meant https://a.aliexpress.com/_mqqh5Nl

closed 2025-04-08 by judge2005