judge2005/EleksTubeIPS #36: [Feature suggestions] Dimming at night, configurable date code

freakyuser-osm, 2025-04-11:
Hello @judge2005,

thanks for your great project!
This was the only reason why I bought an IPSTube, and I'm really enjoying it.

I have two feature ideas, which I think would be a nice addition:

1. Time dependent dimming of screen brightness
I have the clock running during my normal awake time. To be able to read the clock during the day, max. brightness is necessary. But at night lower brightness would be nice.
It would be great to be able to set two different brightness levels. Maybe based on the sunrise and sunset times from the openwheathermap API.

2. Configurable date code
The date code displayed on the weather screen (in time + weather mode) is currently in MM-DD-YYYY. It would be nice to be able to change the order (e.g. DD-MM-YYYY) depending on the local custom.

Maybe these ideas are interesting for you as well.

Regards, freakyuser

judge2005, 2025-04-11:
You can do both of those things:

1. Use the ‘display on’ slider to set the on/off times. Set the ‘off display’ to ‘dimming’.
2. Set the display to ‘date’. An extra control will appear that lets you set the date format. At least I assume that the date in the weather display also follows that. If not that would be a bug.

freakyuser-osm, 2025-04-11:
Thanks for your fast response!

1. Good point. This would work as well. But then it isn't possible to fully turn off the clock during late night hours at the same time.
Would be nice to have the dimming after sunset plus being able to have the display turned during a certain time range.

2. Indeed, didn't spot this setting. Works like a charm!