judge2005/EleksTubeIPS #39: Error 500 on trying to access new Network page

retrozool, 2025-04-12:
Hey there, 

like the title suggest, I am getting a Error Loading Page on the UI (and an underlying Error 500) from my NovelLife SE clock.
I updated from 1.6.6 to 1.6.7 via the installation web site and kept all my settings. 

Restarted the clock multiple times of course and it also seems that the webinterface stops responding after sometime when the error happened. 

If you need any more information or have an idea how to fix it, just let me know.

judge2005, 2025-04-12:
Did you do the full install (first button)? You don't need to check the erase all box.

retrozool, 2025-04-13:
No I didnt - I was absolutely sure it said when I have 1.6.6 I can use the second option as well (Firmware only)
But it seems I cant use the first option again now as it doesnt give me the option to update as I am already on 1.6.7 now

retrozool, 2025-04-13:
I now reflashed everything with the Adafruit Web Tools and that seems to have done the job. So I will close thata gain and sorry for bothering you with a Layer 8 error :D

closed 2025-04-13 by retrozool