judge2005/EleksTubeIPS #40: Problem with flashing.

MadNBG, 2025-04-14:
Hello,

maybe a beginner's qestion...
I got an 401.
when I use the webflasher (Nixies.us), everything is fine.
I then can flash the program - everything OK

But - if I flash the filesystem, /app.html is not found.

I first build the program, 
then I build the filesystem - totally untouched.

No problems while uploading.

Nixie.us ist the only way for me to get it work.

PlattformIO, esptool.exe, esp-webflasher - no Website.

What do I make wrong while flashing the filesystem? Where can I check the problem?

judge2005, 2025-04-14:
I assume you are using platformio. If so, when you build the binary it should also prepare (but not build) the filesystem - so it will place several gzipped files in /data. Then you have separately build the file system binary, which should be an option in the platformio tasks, and then flash that.

I am using platformio in vscode, so perhaps it is different how you are doing it.

The code that builds the gzipped files is in .build_web.py. It requires that nodejs and npm is installed. Platformio should be invoking that before it builds the regular binary because .build_web.py is declared as a pre script in platformio.ini

judge2005, 2025-04-14:
OK. It isn't doing the web build. Do you have [node](https://nodejs.org/en) and npm installed?

MadNBG, 2025-04-14:
Maybe wrong ones? I thought platformio /VS pulls all depencies and requirements?!

[image]

judge2005, 2025-04-14:
You've installed vscode extensions for nodejs and npm, but I wouldn't expect that to install nodejs and npm - only extensions that understand them. I could be wrong though.

The Import("env") line imports platformio environment variables. It is not a python import statement. https://community.platformio.org/t/post-script-import-env-not-working/9840

MadNBG, 2025-04-16:
Thanks for your fast answer. 
I did an install on a clean PC. I installed Python, Nodejs, VScode and Platformio. 
The problem is still existing.
Maybe  the problem occours, because I installed for all users (as admin) and using VScode as a normal user. Maybe relevant data is in the admin's profile (windows) and not in my actual user.. Puuh... I will dig on that. 
I think, I can close this issue - it is - as I thought - a newbe or a OSI-layer 8 problem. Most times, such problems are easy to solve, if knowing where to klick...but, all I read says, "it works". So, it's my really personal problem. 

Thanks for your energy answers - really fast and really helpful. 
So, greetings from Germany!

closed 2025-04-16 by MadNBG

judge2005, 2025-04-16:
Ok, but a likely workaround is for you to manually copy the contents of .../web/src to .../data - maintaining the directory structure.

The nodejs build combines and compresses stuff, but that is probably not necessary - most of it is already compressed so you would only lose a little space in the filing system.

MadNBG, 2025-04-16:
### Solved! 
 (I guess).
After fumbling around, the error has gone. I'm not 100% shure, but **your link** made me deactiving the **Pylance** extension and now it builds complete!