judge2005/EleksTubeIPS #41: Networking and NTP configuration

sbssmia, 2025-04-19:
Hey guys, thought I would drop By and leave my two cents Just uploaded the project works great, but it would be really cool if you could add in the GUI The ability to set static IP / gateway Settings last but not least to be able to set a static NTP server.  I myself run everything local yes I know you could do this in pie hole with redirects and stuff like that But it would really be nice to have that in the GUI keep up the great work love the project. I have a couple clocks thank you to all the contributors

judge2005, 2025-04-23:
That’s possible yes. Allowing you to specify a specific ntp server is easier than static IP allocation. Not sure when I’ll be able to implement them though.

The clock uses pool.ntp.org so you could use that name for your own NTP server.

Anx2k, 2025-05-07:
Another option would be to just get the NTP server from DHCP - if you enable CONFIG_LWIP_DHCP_GET_NTP_SRV, then you'll get a callback dhcp_set_ntp_servers with the NTP servers to use.

judge2005, 2025-11-30:
> Hey guys, thought I would drop By and leave my two cents Just uploaded the project works great, but it would be really cool if you could add in the GUI The ability to set static IP / gateway [...]

A bit late, but is this the kind of thing you had in mind?

[image]

Anx2k, 2025-11-30:
I wouldn't have only static configuration, so like this as a static config and then a button toggle between dhcp and static would be ideal.  And you don't have the NTP server IP specified here - so I would add an edit for that.

agodng, 2025-12-19:
IPSTube H401 - I needed three clarifications

Hi everyone, and congratulations to the author for the wonderful work. I have an IPSTube H401 with the latest original firmware (if I try to update from the app, it says it's the latest)

1. I wanted to know if I can set a 63-character Wi-Fi passkey with the new firmware?

2. If I decide to update the firmware, how can I back up the previous one in case I want to go back?

3. How do I tell if mine has "hardware dimming"? 
 I can easily control the brightness from the official app, this helps me decide which firmware to choose.

Thanks in advance to anyone who can clarify these points.

judge2005, 2025-12-19:
Yes, you should be able to use a 63 character WiFi password

You can follow the procedure [here](https://github.com/judge2005/EleksTubeIPS/wiki/Installation#backup) to back up the current firmware. However all of the releases are also [here](https://github.com/judge2005/EleksTubeIPS/releases), so you can install any version from there.

I don't know of any way to check if your clock supports hardware dimming other than to just try it.

kedube, 2026-11-02:
https://github.com/judge2005/EleksTubeIPS/pull/97 will address this issue.

closed 2026-11-07 by judge2005