judge2005/EleksTubeIPS #50: Screens not powering off with Blank setting

trustoryla, 2025-07-30:
When set to Blank, both the clock scheduling for downtime as well as the Screensaver will not power down the displays. Instead the displays freeze the current image. The Dim and Matrix settings work fine.

Daedelos, 2025-11-16:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

judge2005, 2025-11-18:
OK. Once more with feeling! Install the latest firmware for IPSTube non-hardware-dimming version. The 'firmware-only' install will be fine.

judge2005, 2025-11-19:
> The above issue is with the v1.8.1 firmware for IPSTube non-hardware-dimming version.

Yes. I fixed it in 1.8.2, which is on my website and I updated it in the 1.8.1 release notes, the .bin for that firmware is now at 1.8.2.