judge2005/EleksTubeIPS #56: Random Hanging

Daedelos, 2025-09-16:
After a (seemingly) random number of hours/days my IPSTUBE H401 V2 will hang. Typically not displaying the time but displaying a static backlight pattern.

Looking at the serial output I am seeing this repeated

09:31:05.471 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
09:31:05.507 -> invalid header: 0xffffffff
09:31:05.594 -> invalid header: 0xffffffff
09:31:05.639 -> invalid header: 0xffffffff
09:31:05.685 -> invalid header: 0xffffffff
09:31:05.685 -> invalid header: 0xffffffff
09:31:05.729 -> invalid header: 0xffffffff
09:31:05.807 -> invalid header: 0xffffffff
09:31:05.807 -> ets Jul 29 2019 12:21:46

I have tried numerous power supplies to no avail. My USB power monitor show the clock drawing 0.4A at 5V thus needing 2W. All the power supplies are well above this.

I had the identical issue on my H401 V1 clock and purchased a newer H401 V2 clock thinking possibly the V1 clock had a hardware fault. Sadly they both exhibit the same behaviour.

Using their original firmware both work without fault.

Both were flashed using the webpage method

Daedelos, 2025-09-16:
I've used 1.6.7 and 1.7.0 - both exhibited the same issue.

Daedelos, 2025-09-17:
Leaving it attached to the serial port overnight it didn't hang but produced:

`setup() running on core 1
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0xffffffff,len:-1
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x7f7d791f,len:1330993487
1150 mmu set 00010000, pos 00010000
1150 mmu set 00020000, pos 00020000
1150 mmu set 00030000, pos 00030000
1150 mmu set 00040000, pos 00040000
1150 mmu set 00050000, pos 00050000
1150 mmu set 00060000, pos 00060000
1150 mmu set 00070000, pos 00070000
1150 mmu set 00080000, pos 00080000
1150 mmu set 00090000, pos 00090000
1150 mmu set 000a0000, pos 000a0000
1150 mmu set 000b0000, pos 000b0000
1150 mmu set 000c0000, pos 000c0000
1150 mmu set 000d0000, pos 000d0000
1150 mmu set 000e0000, pos 000e0000
1150 mmu set 000f0000, pos 000f0000
1150 mmu set 00100000, pos 00100000
1150 mmu set 00110000, pos 00110000
1150 mmu set 00120000, pos 00120000
1150 mmu set 00130000, pos 00130000
1150 mmu set 00140000, pos 00140000
1150 mmu set 00150000, pos 00150000
1150 mmu set 00160000, pos 00160000
1150 mmu set 00170000, pos 00170000
1150 mmu set 00180000, pos 00180000
1150 mmu set 00190000, pos 00190000
1150 mmu set 001a0000, pos 001a0000
1150 mmu set 001b0000, pos 001b0000
1150 mmu set 001c0000, pos 001c0000
1150 mmu set 001d0000, pos 001d0000
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x006bd200,len:-2013257536
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x1cce68e1,len:-422537458
ets Jul 29 2019 12:21:46
`

judge2005, 2025-09-17:
That's a lot of watchdog timer resets and it seems the 'hanging' is when it is stuck in a boot loop.

I'll poke around. There are a few places where the IPSTube firmware is different from the other firmwares, so could be a software issue, but at the moment I can't rule out a hardware issue either, especially given the boot loop that causes the hanging.

judge2005, 2025-09-27:
I've made a couple of little changes. Mine has been running fine for about 5 days now - previously it would crash after a day or two. I'll leave it for a few more days then if it still hasn't crashed I'll make another release.

Daedelos, 2025-09-28:
Great, thanks for looking into it.