judge2005/EleksTubeIPS #58: IPSTube H401

CologneMike, 2025-09-30:
Hello, I can't flash my ISPTube H401. I keep getting this error in the terminal, and the browser always times out.

esptool.py --chip esp32 --port /dev/cu.usbserial-14110 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 8MB 0x1000 bootloader8MB.bin 0x8000 partitions8MB.bin 0xe000 boot_app0.bin 0x10000 firmware-ipstube-1.7.1.bin 0x180000 littlefs8MB-1.7.1.bin

esptool v5.1.0
Connected to ESP32 on /dev/cu.usbserial-14110:
Chip type:          ESP32-D0WDR2-V3 (revision v3.1)
Features:           Wi-Fi, BT, Dual Core + LP Core, 240MHz, Vref calibration in eFuse, Coding Scheme None
Crystal frequency:  40MHz
MAC:                e4:65:b8:cd:2b:08

Stub flasher running.
Changing baud rate to 460800...
Changed.

Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00179fff...
Flash will be erased from 0x00180000 to 0x007effff...
SHA digest in image updated.
Wrote 17536 bytes (12201 compressed) at 0x00001000 in 0.5 seconds (273.9 kbit/s).
Input MD5: f46367a8a72b9461591b24b9702d46b6
Flash MD5: c5688fd91caf9b40ab5172e856c0c19b

Hard resetting via RTS pin...

A fatal error occurred: MD5 of file does not match data in flash!

judge2005, 2025-10-01:
Try doing a full install of the latest version I have just released. You should do a full install as the bootloader has changed. This changes flash mode from dio to dout which seems to have fixed some corruption issues.

CologneMike, 2025-10-01:
A full install via the web browser doesn't work either; the regular version gives a 404 error, and the dim version gives a timeout error during installation. I've also tried holding down the button during installation, but unfortunately, it didn't work. The backup I made previously doesn't work either; it gives the error "Magic Byte is FF and must be F9."

judge2005, 2025-10-01:
Just to check, you need to unplug the clock, hold down the button while it is unplugged and then plug it back in while still holding down the button, then release the button and then do the install. I.e. you keep the button pressed while you plug the clock in and then release it after it is plugged in.

Also, do you have anything else plugged in to the computer using USB? If so you need to make sure you select the correct USB port for the install.

closed 2026-01-04 by judge2005